blikvm/blikvm #177: No web interface on Blikvm v4

singspeal1, 2024-08-06:
Hi Support,

I can ping the device but when I put ip address on google chrome. It doesn't show up anything. 

Tks
Kevin

ThomasVon2021, 2024-08-06:
use http not https

singspeal1, 2024-08-06:
I am not using https. Even I can't telnet ipaddress:80 but I can telnet ipaddress:8008

ThomasVon2021, 2024-08-06:
Can you use console port to debug
https://wiki.blicube.com/blikvm/en/BliKVM-v4-guide/#serial-console-access-via-the-5v-port
```
ps -aux | grep link
```
kvm-link is the web soft.

singspeal1, 2024-08-06:
I can use putty to  access the device.

singspeal1, 2024-08-06:
[image]

singspeal1, 2024-08-06:
[image]

singspeal1, 2024-08-06:
> Can you use console port to debug https://wiki.blicube.com/blikvm/en/BliKVM-v4-guide/#serial-console-access-via-the-5v-port
> 
> ```
> ps -aux | grep link
> ```
> 
> kvm-link is the web soft.

sorry I quite dont' get your point. Please advice

ThomasVon2021, 2024-08-06:
```
systemctl status kvmd-web
```

ThomasVon2021, 2024-08-06:
https://github.com/ThomasVon2021/blikvm/tree/master/package/kvmd-web

kvm-link is the web process，If this program doesn't start, it definitely won't be able to open the website. First, check the status to see if you can find the reason for the startup failure. I think flashing the system should be able to restore it.

singspeal1, 2024-08-06:
[image]

singspeal1, 2024-08-06:
Please instruct me how to restore it

ThomasVon2021, 2024-08-06:
https://wiki.blicube.com/blikvm/en/flashing_os/

singspeal1, 2024-08-06:
do u have video to guide to do it step by step. I dont' feel comfortable to do it without very clear guideline
BTW, how can I do 1 to 1 replacement?

ThomasVon2021, 2024-08-06:
There is no video, the problem is very clear. The kvm-link is damaged. You can skip flashing the system and download the latest package of kvm link from GitHub release. Then replace the kvm link program under/usr/bin/blikvm, and restart it to restore normal operation.
https://github.com/ThomasVon2021/blikvm/releases

ThomasVon2021, 2024-08-06:
There should be no problem with the hardware, just replace the software. Please don't worry too much. If it is not fixed in the end, you can return it for a replacement. Please feel free to try it out.

ThomasVon2021, 2024-08-06:
or you can join discord, uncertain steps can be communicated in a timely manner

singspeal1, 2024-08-06:
[image]
which version that I need to download

ThomasVon2021, 2024-08-06:
Can you join Discord? I can send it directly to you in discord, but communication here is too slow.
the release-h616-v4.tar.gz, 
After decompression, there is a kvm-link program inside, just replace it with kvm-ink in/usr/bin/blikvm.

singspeal1, 2024-08-06:
I joned Discord. My user name is singspeal1

closed 2024-08-12 by ThomasVon2021