blikvm/blikvm #178: Can you tell me how to use mouse API?

z0983357347, 2024-08-09:
Where can I find the mouse API in this document? I didn't find.
Or is there someone who can advise on how to use mouse API?
This document didn't find it
https://wiki.blicube.com/blikvm/zh/dev_api/#_2

ThomasVon2021, 2024-08-12:
The latest version to be released has been modified, so the document has not been updated temporarily. It is expected to be updated after the new version is released. The current released version of the mouse is using WS

z0983357347, 2024-08-13:
Thank you for your response.

m50S79sM6SRNp8Jn, 2024-10-13:
What exactly is your use case?

m50S79sM6SRNp8Jn, 2024-10-23:
Please help us understand your use case. Thank you.

ThomasVon2021, 2025-04-22:
https://doc.apipost.net/docs/detail/371d463f8cec000?target_id=283fd94d7cd156
After v1.5.9 version, you can use the mouse api.