blikvm/blikvm #179: Apt Update -> Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)

dfgshdsfh, 2024-08-17:
Just got a BliKVM v4 in, set it up and updated it, which seemed to mostly work but also throws a curious error message

This is the output after having already updated the system once

```
root@mangopimcore:~# apt update
Get:3 http://deb.volian.org/volian scar InRelease [99 B]
Err:3 http://deb.volian.org/volian scar InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Hit:1 https://ryanfortner.github.io/box64-debs/debian ./ InRelease
Hit:4 http://ports.ubuntu.com jammy InRelease
Get:5 http://ports.ubuntu.com jammy-security InRelease [129 kB]
Hit:2 http://armbian.lv.auroradev.org/apt jammy InRelease
Get:6 http://ports.ubuntu.com jammy-updates InRelease [128 kB]
Hit:7 http://ports.ubuntu.com jammy-backports InRelease
Reading package lists... Done
E: Failed to fetch http://deb.volian.org/volian/dists/scar/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: The repository 'http://deb.volian.org/volian scar InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

If you navigate to `http://deb.volian.org/volian/dists/scar/`, you'll find the following message

```
Don't use scar, use dists/nala

https://gitlab.com/volian/volian-archive/-/raw/main/install-nala.sh
```

Here is the output of `cat /etc/apt/sources.list.d/nala.list`
```
deb [signed-by=/usr/share/keyrings/nala.gpg] http://deb.volian.org/volian/ scar main
```

Should I just replace "scar" with "nala"

ThomasVon2021, 2024-08-17:
[image]

closed 2024-08-25 by ThomasVon2021

RojasNet, 2024-08-26:
Thanks was also having this issue and deleting box64.list and nala.list fixed it.