blikvm/blikvm #180: 1.4.1 - BliKVM Switch not working

jk-andersen, 2024-08-25:
Updated to 1.4.1 without any problems.

I setup my BliKVM-Switch 1 in the config. Log:
`
[2024-08-25T07:26:58.768] [INFO] file - http api request POST /api/device by admin
[2024-08-25T07:26:58.831] [INFO] file - http api request POST /api/switch/getlist by admin
[2024-08-25T07:26:58.854] [INFO] file - http api request POST /api/msd/images by admin
[2024-08-25T07:26:58.933] [INFO] file - http api request POST /api/video/config?action=get by admin
[2024-08-25T07:26:58.961] [INFO] file - http api request POST /api/switch/state?module=BliKVM_switch_v1 by admin
[2024-08-25T07:26:58.966] [INFO] file - KVMDBliSwitchV1 init success, path: /dev/ttyUSB0, name: BliKVM_switch_v1`

The Webinterface shows this and i cannot switch the channels:

[image]

ThomasVon2021, 2024-08-25:
What hardware version are you using?

jk-andersen, 2024-08-25:
Sorry: BliKVM v4 Allwinner

ThomasVon2021, 2024-08-25:
Do you try use web ui enable?
 In /usr/bin/blikvm/package.json switch control no useful now.
 
[image]

ThomasVon2021, 2024-08-25:
developer tool?

ThomasVon2021, 2024-08-25:
Is using web control effective? The test on my end is effective.

jk-andersen, 2024-08-25:
I did that now. It started working after I changed the "channelnames" back to OOTB (channel_1,...).
Ít also freezes the webui at my installation (ssh still up, but webserver not responding). But the switch now works and changes the channel

ThomasVon2021, 2024-08-25:
Has the web stopped working?

jk-andersen, 2024-08-26:
The Webinterface gets ujnresponsiove and does not even respond to pings when I change the channels.
I tested it with the 1.4.2 release now. Here the Switch works directly after enabling in the Webinterface, but also here - When I change the channel 2 times the WebUI breaks. Only a reboot via ssh recovers this.

ThomasVon2021, 2024-08-27:
Thank you for your feedback. I will proceed with testing.

ThomasVon2021, 2024-08-27:
What did you change the channel name to? Can you provide me with it? Also, could you please list the specific steps? I repeatedly switched without reproducing.

jk-andersen, 2024-08-27:
[image]

This is my config.
The log from the webui before "switching":

```
-- INFO  [97.361    stream] -- Using IO method: MMAP

[2024-08-26T11:17:07.736] [TRACE] file - video API stderr: -- INFO  [97.371    stream] -- Requested 5 device buffers, got 5

[2024-08-26T11:17:07.743] [TRACE] file - video API stderr: -- INFO  [97.378    stream] -- Capturing started
-- INFO  [97.378    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [97.378    stream] -- Using JPEG quality: encoder default
-- INFO  [97.378    stream] -- Creating pool JPEG with 1 workers ...

[2024-08-26T11:17:07.745] [TRACE] file - video API stderr: -- INFO  [97.378    stream] -- Capturing ...

[2024-08-27T09:51:35.981] [INFO] file - http api request POST /api/switch/getlist by admin
[2024-08-27T09:51:36.004] [INFO] file - http api request POST /api/device by admin
[2024-08-27T09:51:36.031] [INFO] file - http api request POST /api/msd/images by admin
[2024-08-27T09:51:36.105] [INFO] file - http api request POST /api/video/config?action=get by admin
[2024-08-27T09:51:36.112] [TRACE] file - video API stderr: -- INFO  [81362.962      http] -- HTTP: NEW client (now=1): [10.0.0.70]:54784, id=d738729cb5586504

[2024-08-27T09:51:36.132] [INFO] file - WebSocket Client connected, total clients: 1
[2024-08-27T09:51:36.148] [INFO] file - http api request POST /api/switch/state?module=BliKVM_switch_v1 by admin
[2024-08-27T09:51:36.196] [INFO] file - http api request POST /api/msd/state by admin
```

Can I somehow see the log on ssh? I could invsstsigate there.

ThomasVon2021, 2024-08-27:
Yes, log is in /mnt/tmp/logs/
https://github.com/ThomasVon2021/blikvm-web-server/blob/master/src/modules/kvmd/switch/kvmd_switch_v1.js
I found that there is a problem with the serial port reading of the switch, and I am currently investigating whether this will cause the program on your end to crash.
btw, the web process is server_app
you can use "ps -aux | grep server_app" to check.

ThomasVon2021, 2024-08-27:
Can you let me remotely access your device? That way, debugging will be much simpler

ThomasVon2021, 2024-08-27:
[image]
[image]

If you can ssh, use tail -f /mnt/tmp/logs/app.log, maybe not applog, you need to check which log is the lasted.
Let's see if we can make any discoveries.

jk-andersen, 2024-08-27:
Ok. My debugging steps:

Verified the switch is working and connected correctly. It switches with the Hardware buttons on device. I also restarted it.

- deleted app.log
- restarted the Blikvm
- Access via WebUI
- Switched on the switch toggle.

WebUI becomes unresponsive

applog:

```
[2024-08-27T11:43:49.723] [INFO] file - Http Api started at http://localhost:80, state: RUNNING
[2024-08-27T11:43:49.726] [INFO] file - WebSocket Api started at ws://localhost:80, state: RUNNING
[2024-08-27T11:43:49.728] [INFO] file - tus server listening at http://0.0.0.0:10002
[2024-08-27T11:43:49.739] [INFO] file - ./lib/hid/enable-gadget.sh
[2024-08-27T11:43:49.876] [INFO] file - All services have been started.
[2024-08-27T11:43:49.971] [WARN] file - Notification WebSocket is not initialized.
[2024-08-27T11:43:50.014] [TRACE] file - video API stdout: network ok

[2024-08-27T11:43:50.020] [INFO] file - HID start success
[2024-08-27T11:43:50.024] [INFO] file - video API started, state: RUNNING
[2024-08-27T11:43:50.046] [INFO] file - kvmd-main API started, state: RUNNING
[2024-08-27T11:43:54.877] [INFO] file - ATX API started, state: RUNNING
[2024-08-27T11:43:55.865] [TRACE] file - video API stdout: Board type: V4_H616

[2024-08-27T11:43:56.084] [TRACE] file - video API stdout: find support JPEG video divice: /dev/video1

[2024-08-27T11:43:56.130] [TRACE] file - video API stderr: -- INFO  [97.203      main] -- Starting PiKVM uStreamer 5.38 ...
-- INFO  [97.203      main] -- Using internal blank placeholder

[2024-08-27T11:43:56.135] [TRACE] file - video API stderr: -- INFO  [97.204      main] -- Listening HTTP on [0.0.0.0]:10004
-- INFO  [97.204    stream] -- Using V4L2 device: /dev/video1
-- INFO  [97.204    stream] -- Using desired FPS: 30
-- INFO  [97.205      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [97.205    stream] -- Device fd=8 opened
-- INFO  [97.206    stream] -- Using input channel: 0

[2024-08-27T11:43:56.152] [TRACE] file - video API stderr: -- INFO  [97.225    stream] -- Using resolution: 1920x1080
-- INFO  [97.225    stream] -- Using format: MJPEG

[2024-08-27T11:43:56.174] [TRACE] file - video API stderr: -- INFO  [97.246    stream] -- Using HW FPS: 30
-- ERROR [97.246    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [97.246    stream] -- Using IO method: MMAP

[2024-08-27T11:43:56.184] [TRACE] file - video API stderr: -- INFO  [97.257    stream] -- Requested 5 device buffers, got 5

[2024-08-27T11:43:56.193] [TRACE] file - video API stderr: -- INFO  [97.266    stream] -- Capturing started
-- INFO  [97.266    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [97.266    stream] -- Using JPEG quality: encoder default
-- INFO  [97.266    stream] -- Creating pool JPEG with 1 workers ...

[2024-08-27T11:43:56.195] [TRACE] file - video API stderr: -- INFO  [97.266    stream] -- Capturing ...

[2024-08-27T11:44:03.100] [INFO] file - http api request POST /api/switch/getlist by admin
[2024-08-27T11:44:03.124] [INFO] file - http api request POST /api/msd/images by admin
[2024-08-27T11:44:03.200] [INFO] file - http api request POST /api/video/config?action=get by admin
[2024-08-27T11:44:03.219] [INFO] file - http api request POST /api/device by admin
[2024-08-27T11:44:03.231] [TRACE] file - video API stderr: -- INFO  [104.136      http] -- HTTP: NEW client (now=1): [10.0.0.70]:61514, id=9a589b64c2338e89

[2024-08-27T11:44:03.251] [INFO] file - WebSocket Client connected, total clients: 1
[2024-08-27T11:44:03.268] [INFO] file - http api request POST /api/switch/state?module=BliKVM_switch_v1 by admin
[2024-08-27T11:44:03.272] [INFO] file - KVMDBliSwitchV1 init success, path: /dev/ttyUSB0, name: BliKVM_switch_v1
[2024-08-27T11:44:03.317] [INFO] file - http api request POST /api/msd/state by admin
```

Check of the server_app:

```
blikvm@mangopimcore:/mnt/tmp/logs$ ps -aux | grep server_app
root        2095  4.7  6.5 2819696 66100 ?       Ssl  11:43   0:08 /mnt/exec/release/server_app
blikvm      2516  0.0  0.1   6024  1992 pts/0    S+   11:46   0:00 grep --color=auto server_app
```

I passed through the blikvm via traeffik for debugging - Will send you the details in a private message.

ThomasVon2021, 2024-08-27:
Do you have Discord? Can you contact me on Discord?
My name is Vons
[image]

closed 2024-08-28 by ThomasVon2021