blikvm/blikvm #184: Upgrade failure BliKVM v4 Allwinner

benalexau, 2024-09-06:
As part of testing the fix for #79 I attempted to upgrade a BlikKVM v4 Allwinner.

First I updated the OS via SSH:

```
apt-mark hold linux-dtb-edge-sunxi64 linux-image-edge-sunxi64
sudo apt-get update
sudo apt-get upgrade
reboot
```

After rebooting I SSHed in again:

```
sudo -i
cd /
mkdir update
cd update
wget https://github.com/ThomasVon2021/blikvm/releases/download/v1.4.3/release-h616-v4-alpha.tar.gz
tar -zxvf release-h616-v4-alpha.tar.gz 
cd release/
python3 install_release.py --releasepath=./
```

This failed with:

```
bash: install-kvmd-main.sh: No such file or directory
Traceback (most recent call last):
  File "/update/release/install_release.py", line 77, in <module>
    main()
  File "/update/release/install_release.py", line 70, in main
    subprocess.check_output(cmd, shell = True, cwd=gArgs.releasepath )
  File "/usr/lib/python3.10/subprocess.py", line 421, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'bash install-kvmd-main.sh && bash install-ustreamer.sh && bash install-kvmd-hid.sh             && bash install-kvmd-web.sh && bash install-kvmd-msd.sh && cp package.json /usr/bin/blikvm/package.json' returned non-zero exit status 127.
```

The requested `install-kvmd-main.sh` does not exist:

```
# ls -l
total 444
-rwxr-xr-x 1 root root    471 Aug 28 07:13 install-kvmd-web.sh
-rwxr-xr-x 1 root root   2929 Aug 28 07:13 install_release.py
-rwxr-xr-x 1 root root    348 Aug 28 07:13 kvmd-web.service
-rwxr-xr-x 1 root root    243 Aug 28 07:13 package.json
-rw-r--r-- 1 root root    235 Aug 28 07:13 readme.md
drwxr-xr-x 4 root root   4096 Aug 28 07:13 release
-rwxr-xr-x 1 root root 429136 Aug 28 07:13 ustreamer.bin
root@mangopimcore:/update/release# 
```

Any suggestions?

ThomasVon2021, 2024-09-06:
# If you want test the newest alpha version, use:
python3 /opt/bin/blikvm/script/update.py alpha

https://wiki.blicube.com/blikvm/en/update/#introduction

ThomasVon2021, 2024-09-06:
or
python3 install_release.py --releasepath=./ --alpha=true

benalexau, 2024-09-06:
Thanks for the instructions. I followed your [earlier comment](https://github.com/ThomasVon2021/blikvm/issues/184#issuecomment-2333027516), but found `python3 /opt/bin/blikvm/script/update.py` without the `alpha` parameter reports a release not found error. It does work with `alpha` though:

```
$ python3 /opt/bin/blikvm/script/update.py alpha
Board type: BoardType.V4_H616
The latest release tag for blikvm is  v1.4.3
The local version is  Develop-1.2.0
Upgrading  Develop-1.2.0  ==>  v1.4.3
Download package:  release-h616-v4-alpha.tar.gz  please wait...
Download progress: 0.00%
[....]
Download progress: 100.00%
Download progress: 100.00%
release-h616-v4-alpha.tar.gz downloaded to /tmp/kvm_update/release-h616-v4-alpha.tar.gz successfully.
Download release package success, start to install, please wait...
Upgrade successful!
$ reboot
```

I was able to login to the updated web UI, although the issue in #79 remains. Do I need to set a config somewhere to enable relative mouse modes?

ThomasVon2021, 2024-09-06:
UI mouse have the config. Relative mode estimation has not been tested by anyone yet, there may be issues, waiting for your feedback after testing
[image]

closed 2024-09-06 by benalexau