blikvm/blikvm #188: error: `/dev/hidg0` does not exist

mircea-pavel-anton, 2024-09-19:
I have configured some more shortcuts in my `app.json` like this:

```json
...
  "hid": {
    "hidEnable": "./lib/hid/enable-gadget.sh",
    "hidDisable": "./lib/hid/disable-gadget.sh",
    "enable": false,
    "absoluteMode": true,
    "mouseJiggler": true,
    "shortcuts": {
      "Ctrl+Alt+Del": [
        "ControlLeft",
        "AltLeft",
        "Delete"
      ],
      "Alt+Tab": [
        "AltLeft",
        "Tab"
      ],
      "Alt+F4": [
        "AltLeft",
        "F4"
      ],
      "TE01": [
        "ScrollLock",
        "ScrollLock",
        "Digit1"
      ],
      "TE02": [
        "ScrollLock",
        "ScrollLock",
        "Digit2"
      ],
...
      "TE15": [
        "ScrollLock",
        "ScrollLock",
        "Digit1",
        "Digit5"
      ],
      "TE16": [
        "ScrollLock",
        "ScrollLock",
        "Digit1",
        "Digit6"
      ]
    }
  }
```

This allows me to control my 16 port TESmart KVM inputs via the WebUI via the shortcut buttons.

The problem is that it doesn't always work as intended. If i take a look in my log file, I can see these logs whenever I press on any of the buttons:

```
[2024-09-19T21:07:57.229] [INFO] file - http api request POST /api/hid/shortcuts by admin
[2024-09-19T21:07:57.233] [ERROR] file - File /dev/hidg0 does not exist
[2024-09-19T21:07:57.285] [ERROR] file - File /dev/hidg0 does not exist
```

mircea-pavel-anton, 2024-09-19:
Update: this seems to happen even when I activate the on-screen keyboard and press any button:

```
[2024-09-19T21:20:36.184] [ERROR] file - File /dev/hidg0 does not exist
``

m50S79sM6SRNp8Jn, 2024-09-19:
Let's take the switch out of the equation. It will be helpful if you can test without the switch and let us know the result. Thanks.

ThomasVon2021, 2024-09-20:
OK, discussed in Discord, this issue should be easy to solve.

ThomasVon2021, 2024-09-20:
"enable": false,

why do you enable your hid？That will disable keyboard and mouse.

mircea-pavel-anton, 2024-09-21:
Update: The issue was that my KVM does USB emulation on the kb/mouse ports and that was interfering with the mouse commands being sent through. Plugging the BliKVM into one of the hub ports fixed all my issues!

closed 2024-09-21 by mircea-pavel-anton