blikvm/blikvm #189: Mouse not working

mircea-pavel-anton, 2024-09-19:
I have just bought my first BliKVM from AliExpress and it arrived a few days ago. I plugged everything in correctly, I logged into the WebUI and I noticed I can't seem to get my mouse working to a windows host.

I thought I should just update the system and try again. I did update to the latest version (1.4.5), but it still does not seem to work? Not sure how to troubleshoot this. I simply dont get a cursor on the screen at all

ThomasVon2021, 2024-09-20:
I saw your message on Discord, we can continue to communicate on Discord.

mircea-pavel-anton, 2024-09-21:
Update: The issue was that my KVM does USB emulation on the kb/mouse ports and that was interfering with the mouse commands being sent through. Plugging the BliKVM into one of the hub ports fixed all my issues!

closed 2024-09-21 by mircea-pavel-anton