blikvm/blikvm #190: Updated to 1.4.5 - cannot log in to web interface anymore

swingstate, 2024-09-19:
I upgraded through the ssh script from any earlier 1.4.x build. 
Update went fine, rebooted. 
Now, I cannot login with the account defined in the admin.ini anymore. 

Any advice?

mircea-pavel-anton, 2024-09-19:
Is the web UI not showing up or are the credentials just not working? I also ran into this and I just rebooted once more and it came back up

ThomasVon2021, 2024-09-20:
> I upgraded through the ssh script from any earlier 1.4.x build. Update went fine, rebooted. Now, I cannot login with the account defined in the admin.ini anymore.
> 
> Any advice?

Default is reset to admin/admin

swingstate, 2024-09-20:
That was too easy. Thanks Thomas. Is that reset documented somewhere and will that happen with every reset in the new version?

Do we still need the admin.ini?

ThomasVon2021, 2024-09-20:
In version 1.4.6, there will be no reset, you can try it out.
Don't need admin.ini anymore.

closed 2024-09-21 by swingstate