blikvm/blikvm #191: [1.4.5-alpha] SSH Webterminal not working

swingstate, 2024-09-21:
I tried to use the Terminal in the webbrowser but get following error: 

connecting...
Connection failed, Please check your account and password information, then you can refresh and try again.

The log says:

SSH CONNECTION ERROR: All configured authentication methods failed.

Do I need to enable the web terminal somewhere?

ThomasVon2021, 2024-09-21:
You change the ssh username or password, right? If you do it, you need to change the /mnt/exec/release/config/app.json

swingstate, 2024-09-21:
Thanks, are you planing to document these necessary configuration steps in the wiki/docu and would you consider to move away from storing plain text passwords in the configuration files?

ThomasVon2021, 2024-09-21:
Wiki was added. About text passwords, can you open a new issue for it? So When we do it, you will know.

m50S79sM6SRNp8Jn, 2024-09-21:
> Wiki was added. About text passwords, can you open a new issue for it? So When we do it, you will know.

https://github.com/ThomasVon2021/blikvm/issues/192

ThomasVon2021, 2024-09-22:
https://wiki.blicube.com/blikvm/en/ssh/

closed 2024-10-10 by ThomasVon2021