blikvm/blikvm #193: Block ATX commands when using external multiport KVM unless on correct port/device

matthewbarreiro, 2024-09-22:
I don't think this has been requested yet, but apologies if this is a duplicate.

When using an external multi-port KVM switch (like the one previously sold by your team) and ATX together (e.g. BliKVM v4), it's very easy to accidentally send ATX power commands to the wrong device. It would be really nice to be able to define which port the ATX cable is connected to, and have the web interface either block clicking or hide the ATX buttons unless you are connected to the correct device.

m50S79sM6SRNp8Jn, 2024-09-23:
Which BliKVM software version are using now?

mrulke, 2024-10-18:
i am using 1.4.8  and using the hatv3 with piv4

mrulke, 2024-10-18:
could you look at integrating this?
https://github.com/zappanaut/pikvm-usb-atx-ctrl

ThomasVon2021, 2025-02-08:
https://wiki.blicube.com/blikvm/en/bliswitch-v2-guide/

Maybe switch-v2 can solve your questions？

matthewbarreiro, 2025-02-10:
> https://wiki.blicube.com/blikvm/en/bliswitch-v2-guide/
> 
> Maybe switch-v2 can solve your questions？

I saw that, and it's nice, but doesn't really solve this issue. This issue is about ensuring that when using the switch v1 (or any similar hardware that doesn't switch ATX), you can only send ATX commands when the BliKVM is displaying the feed for the device that actually has ATX connected.

mrulke, 2025-03-13:
Yeah i already have the v1 switch

> https://wiki.blicube.com/blikvm/en/bliswitch-v2-guide/
> 
> Maybe switch-v2 can solve your questions？

ThomasVon2021, 2025-04-22:
```
  "atx": {
    "controlSockFilePath": "/var/blikvm/atx.sock",
    "stateSockFilePath": "/dev/shm/blikvm/atx",
    "power_on_delay": 500,
    "power_off_delay": 5000,
    "switch_channel": -1;
  },
```
Maybe I can add a switch_channel in atx config:
1. If switch_channel=-1, send atx command directly.
2. If switch_channel != -1, only send atx command when active switch channel = switch_channel

m50S79sM6SRNp8Jn, 2025-04-22:
In the new GUI, I was thinking about “experimental” features. It’s a feature but it’s highlighted as “experimental “. It’s up to the user to try it or not.