blikvm/blikvm #194: auto. turn off LCD display on v4

jhaveDK, 2024-10-03:
A feature req. for v4 is a add a option to turn the display on boot for like 30 sec. and then turn it off. The only way to get i back alive is to push on of the 2 buttons or a reboot.

ThomasVon2021, 2024-10-09:
https://wiki.blicube.com/blikvm/en/OLED-display/
The latest version has this feature, you can modify restart_sthow_time

jhaveDK, 2024-10-22:
I have now tested it in version 1.4.8 with the settings 

`"oled":{
        "oled_enable": 0,
        "restart_show_time": 300,
        "interval_display_time": 5
    },`
    
 But its dont give the result with the screen on for 30 seconds and then its turn off and its only possible to wake i with a button push.

m50S79sM6SRNp8Jn, 2024-10-29:
> its dont give the result with the screen on for 30 seconds

When you have a moment can you please clarify the behaviour you are seeing and what you are expecting?

jhaveDK, 2024-10-29:
I have added a time diagram to show what i mean, when its boot the LCD is on for 30 sec. and when one of the buttons is pushed its on for 30 sec. 

[image]

m50S79sM6SRNp8Jn, 2024-10-29:
What behaviour are you expecting?

m50S79sM6SRNp8Jn, 2024-10-29:
Ok, will take a closer look at it.

m50S79sM6SRNp8Jn, 2024-10-29:
> Its turn the screen off, other venders like unifi use this model for there devices.

Hi @jhaveDK,

When you have a moment, how is your comment different from https://github.com/ThomasVon2021/blikvm/issues/204 ?

jhaveDK, 2024-10-29:
> > Its turn the screen off, other venders like unifi use this model for there devices.
> 
> Hi @jhaveDK,
> 
> When you have a moment, how is your comment different from #204 ?

#204 is about the backlight, this is about the LCD panel work and how its turned on and off by a timer.

m50S79sM6SRNp8Jn, 2024-11-01:
@jhaveDK after analysis, here's a breakdown of the requirements for the BliKVM v4 Allwinner LCD control.
When you have a moment, please review and let us know any adjustment(s).

Principle
To avoid LCD burn-in, limit the LCD's on-time whenever possible.

Scenarios
**1) Initial Always-On Mode (New Device Setup)**:

Purpose: For users who are new to the BliKVM v4 device, always-on mode will allow them to familiarize themselves with the interface without worrying about time limits.

Details: After a new device setup, the LCD remains always on for an initial period, until manually disabled in the GUI.
Transition: After this initial setup phase, the device will automatically switch to Scenario 1 and Scenario 2 behaviors to conserve the display.

**2) Scenario: User Not Present**:

Purpose: When the user isn’t nearby, the LCD should remain off to prevent unnecessary wear.
Behavior: The display stays off entirely in this scenario.

**3) Scenario: User Present**:

Purpose: When the user is nearby and needs to view information, the LCD will turn on briefly.

Behavior:
On Boot: The LCD turns on for n seconds to display initial information, then automatically turns off.
On Button Press (sw1): Pressing sw1 turns on the LCD for n seconds. If the LCD is already on, pressing sw1 resets the timer, giving the user another n seconds.

Parameter Adjustment:

The timer duration n should be user-configurable in the GUI, with an initial default of 30 seconds. However, for new users, this default can be set to a longer time (e.g., 60 seconds or more) until they are comfortable with the device. After familiarization, users can adjust it to 30 seconds or another preferred value.

Summary

This setup prevents LCD burn-in while providing an initial learning phase for new users. The adjustable timer allows users to fine-tune display timing as they grow accustomed to the device.

jhaveDK, 2024-11-01:
Its sounds correct.

m50S79sM6SRNp8Jn, 2024-11-01:
> Its sounds correct.

Thanks for your confirmation. Once there is a significant update, will report back.

m50S79sM6SRNp8Jn, 2024-11-09:
Parameters are controllable in JSON structure.
In a later release we will have add a GUI for it.

closed 2024-11-16 by m50S79sM6SRNp8Jn