blikvm/blikvm #195: Special characters are not sent

Yeocti, 2024-10-06:
When using a french keyboard (AZERTY) special character such as @ or # are not sent.
On french keyboards Alt-Gr is used to send those special characters.

An alternative is to use Left Ctlr+Left Alt instead of Alt Gr however it doesn't work either.

This issue was probably introduced in matrix release and was not present before.

BliKVM v4 running 1.4.7 alpha.
Using Firefox 130.0.1 on Windows 10 to access BliKVM GUI.
Controlled machine is running Linux with fr keymap.

ThomasVon2021, 2024-10-11:
Does your windows is eng keymap？

ThomasVon2021, 2024-10-11:
Now I have windows and ubuntu pc. Can you tell me how to reproduce it? I have no experience using keyboards from other countries.

Yeocti, 2024-10-11:
I'm using bliKVM from windows 10. Keymap is french (azerty keyboard).
Controlled machine is linux with fr keymap.

Azerty keyboard looks like:
[image]

In order to send special characters such as # [ or @, you have to press Alt-Gr + 3 or 5 or 0 (respectively).
So if you want to reproduce, you'd have to add french locale to windows and then set your keyboard to french layout.
Then, when using bliKVM, you'll notice that you're unable to send those special characters (and all others that need the use of Alt-Gr to be sent).

ThomasVon2021, 2024-10-11:
Do I need a real French keyboard to test this?
https://github.com/ThomasVon2021/blikvm-web-client/blob/master/src/utils/virtualKeyboard.js
Now the keycode don't have AltGr, which version do you test it is ok?

Yeocti, 2024-10-11:
> Do I need a real French keyboard to test this?
No. You can just change the layout of your current keyboard.

If you're using windows 10, Select Start > Settings > Time & Language > Language.
Add french as a new language.  Then in keyboard you can choose the french layout.
I might be easier if you use the windows virtual keyboard.

> https://github.com/ThomasVon2021/blikvm-web-client/blob/master/src/utils/virtualKeyboard.js
> Now the keycode don't have AltGr, which version do you test it is ok?

Not sure to understand your question. Using a physical keyboard or the virtual keyboard does the same.
As you pointed AltGr is not mapped in the virtual keyboard. But you can replace AltGr by pressing LeftCtrl and LeftAlt + _special character_ and it does not work either even so Ctrl and Alt are mapped in your virtual keyboard.

What did you use prior to matrix release to capture keyboard input?

ThomasVon2021, 2024-10-12:
[image]
Can you try shortcut like ctrl+alt_del, see if he will take effect？
If this can take effect, it means that the combination of Ctrl+Alt keys is possible. You can test the @ symbol with a shortcut.
The keyboard code is basically the same as before, with no significant changes.

Yeocti, 2024-10-13:
Ctrl+Alt+Del works with the shortcut because whereas it's AZERTY or QWERTY keyboard it's the same physical keys.
I was unable to create a working shortcut to get @.

I found a way to get # without creating a shortcut. However it's a bit strange.
On a QWERTY keyboard to get # you have to press Shift + 3
On a AZERTY keyboard it would be Shift + " to get the same physical keys pressed.

So if on my AZERTY keyboard I press Shift + µ it's like I pressed Shift + " on a QWERTY keyboard.
On the console it's translated as if I pressed Shift + 3 and I get #. That's weird !

I think there is definitely a problem with the way key press are converted in your virtual keyboard.

I then tried to change the keymap on controlled machine to eng.
Doing so i can use my physical keyboard and it works fine.

But the virtual keyboard does not work well. Shift and Caps lock keys have not effects.
If i press Shift or Caps lock I can see the keys changing on the virtual keyboard but when i click on a key it's like Shift/Caps lock was not activated.

ThomasVon2021, 2024-10-14:
virtual keyboard don't support shortcut now.
Do you mean if you use AZERTY keyboard, and target is eng, it is ok?

Yeocti, 2024-10-14:
yes. It's like i was using a QWERTY keyboard...

blikvm, 2025-10-31:
> 这个问题有任何进展吗？我在 2.1.3 alpha 版本上也遇到了完全相同的问题。无法在 Web 管理界面中输入 @ 键（AltGr + 2）。服务器和本地 Windows 11 电脑的键盘布局相同（挪威语）。PiKVM 似乎已经解决了类似的问题（[pikvm/pikvm#375](https://github.com/pikvm/pikvm/issues/375)）。这两个问题有关联吗？

Yes, can you add me to Discord? I am referring to PiKVM's solution. If you can cooperate with debugging, I think it will be much faster

blikvm, 2025-10-31:
I have purchased a German keyboard and will test it as soon as possible.

[image]

blikvm, 2025-11-03:
If anyone can help test, I can give a test version to test it.

blikvm, 2025-11-14:
> Are there any updates to this issue. I am having the exact same problem on 2.1.3 alpha. Unable to get @ (AltGr + 2) in the web-gui. Both server and local windows 11 computer have the same keyboard layout (Norwegian). A similar issue seems to have been resolved for PiKVM ([pikvm/pikvm#375](https://github.com/pikvm/pikvm/issues/375)). Are these issues related?

v2.2.1-alpha may solve AltGr bug, can you test it?

m50S79sM6SRNp8Jn, 2025-11-17:
Sup?

m50S79sM6SRNp8Jn, 2025-11-28:
Thanks for your confirmation.

closed 2025-11-28 by m50S79sM6SRNp8Jn