blikvm/blikvm #198: [Feature] Secure HTTPS Certificate Issuance for BliKVM Behind NAT Using WireGuard and NGINX Proxy

m50S79sM6SRNp8Jn, 2024-10-13:
Problem
BliKVM devices often run in environments without a public IP address (e.g., behind a NAT or firewall), making it impossible to complete Let's Encrypt ACME challenges required for HTTPS certificates. Exposing the BliKVM directly to the public internet is not ideal due to security concerns, such as leaking API keys or exposing sensitive infrastructure.

Proposed Solution
To securely obtain HTTPS certificates for a BliKVM that lacks a public IP address, we can set up a WireGuard VPN tunnel to a publicly accessible VPS and use an NGINX reverse proxy on the VPS to forward Let's Encrypt ACME challenges over the WireGuard tunnel. This allows the BliKVM to appear publicly reachable to Let's Encrypt while remaining safely hidden behind a firewall.

ThomasVon2021, 2025-02-26:
I will test it later.

ThomasVon2021, 2025-03-03:
https://wiki.blicube.com/blikvm/en/wireguard/

closed 2025-03-03 by ThomasVon2021