blikvm/blikvm #213: Secure Blikvm further: whitelisted ip's support

kaiyuan01, 2024-11-08:
**Problem Description**
only allows blikvm to accept requests from a whitelisted ip's to secure kvm further, esp. in port forwarding scenarios.

**Proposed Solution**
Make server code change so that it only allows blikvm to accept requests from a whitelisted ip's to secure kvm further, esp. in port forwarding scenarios.

**Alternatives Considered**
None

**Additional context**
This helps further secure Blikvm

m50S79sM6SRNp8Jn, 2024-11-08:
For external BliKVM exposure we recommend Tailscale.
What would you proposed feature bring additionally, besides multi-layered security?

m50S79sM6SRNp8Jn, 2024-11-08:
I am not opposed to multi-layered security.

kaiyuan01, 2024-11-09:
> Tailscale.

For Tailscale to work, do I have to either install a Tailscale client on my host PC (not an option for me), or put KVM behind a VPN router suporting Wireguard, which is not something I plan on doing.

Any issues with port forwarding? It requires neither of the sw installation, or a VPN enabled router. To secure kvm in this scenario, ip whitelisting is a feature necessary to further protect the system. Hope this makes sense and it should not be difficult to do I think. If you can point me to the server code that handles user requests, I can contribute too.

ThomasVon2021, 2024-11-09:
https://github.com/ThomasVon2021/blikvm-web-server/blob/master/src/server/api/routes.js

m50S79sM6SRNp8Jn, 2024-11-09:
> Any issues with port forwarding?

No issues per se, just to be aware the port forwarding elevates security risks.
Further, there is a need to handle dynamic IP.

You will need
1. VPN + IP Whitelisting (only devices on the VPN with approved IPs can connect)
2. 2FA support
3. Cloudflare integration (to support dynamic IP)

Am I missing anything?

m50S79sM6SRNp8Jn, 2024-11-12:
These are in separate GitHub requests.

ThomasVon2021, 2025-03-12:
Please update to v1.5.7 to test.
https://wiki.blicube.com/blikvm/en/auth/#two-factor-authentication.
If there still have problem, you can reopen the issue.

closed 2025-03-12 by ThomasVon2021