blikvm/blikvm #220: [Feature request] On first deployment/upgrade, update /etc/machine-id so these devices are recognized as unique

bmcminn, 2024-11-18:
**Problem Description**
I've had an issue I detailed in the Tailscale repo https://github.com/tailscale/tailscale/issues/14088 where it seemed that Tailscale was using the same device fingerprint against both BliKVMs I was deploying on two geographically separated networks.

Both units were updated, upgraded, latest firmware, etc. However their `/etc/machine-id` values matched which I think was the root of the issue, since I would try to add one device to my tailnet which worked, and then adding the other device would overwrite the existing tailnet client, and the tailscale logs would just say there as a name update.

**Proposed Solution**
An automated script on first launch would be better to ensure each KVM instance `machine-id` is actually unique, but it would be helpful if the tailscale/VPN docs mentioned you need to ensure the `machine-id` of the device is unique before deploying multiple instances.

```bash
#!/bin/bash
# This script would need to be run with sudo permissions
# sauce: https://unix.stackexchange.com/a/403054

rm -rf /etc/machine-id
rm -rf /var/lib/dbus/machine-id

dbus-uuidgen --ensure=/etc/machine-id
dbus-uuidgen --ensure

reboot
```
And then install tailscale and connect it to your VPN/tailnet

m50S79sM6SRNp8Jn, 2024-11-18:
Nice catch. We will review and any questions we will come back to you. Thanks.

m50S79sM6SRNp8Jn, 2024-11-18:
Which BliKVM hardware device(s) and software you are using?
How many devices?

bmcminn, 2024-11-18:
I'm running two BliKVM PCIe v2 devices plugged into their respective server hosts.

BliKVM release v1.5.0 - https://github.com/ThomasVon2021/blikvm/releases/tag/v1.5.0
OS version: Linux 6.1.21-v8+
Tailscale: v1.76.6

ThomasVon2021, 2024-11-26:
[image]

ThomasVon2021, 2024-11-26:
[image]

closed 2024-11-28 by m50S79sM6SRNp8Jn