blikvm/blikvm #232: [Feature Request] Disable WebUI Authentication

jwgn, 2024-12-10:
**Problem Description**
As I currently use my BLIKVM v4 only in my home network, it would be convenient not having to enter credentials for WebUI access.

**Proposed Solution**
Add an option to disable WebUI authentication entirely

**Alternatives Considered**
None.

**Additional context**
None.

m50S79sM6SRNp8Jn, 2024-12-10:
Do you by any chance use a Yubikey?

jwgn, 2024-12-10:
I use multiple hardware keys, but I do not intend to move them between my devices just to access a WebUI. It would be more complicated than entering the login credentials.

m50S79sM6SRNp8Jn, 2024-12-10:
Would you use a password manager? Would that allow you to automatically login, while using authentication credentials?

jwgn, 2024-12-10:
This is not what I asked for. My request was to disable WebUI authentication entirely.

m50S79sM6SRNp8Jn, 2024-12-10:
True. Was just thinking of an other / near immediate way of dealing with the inconvenience of having to enter credentials for WebUI access.

ThomasVon2021, 2025-01-27:
you can modify /mnt/exec/release/app.json 
find auth, set to false

jwgn, 2025-01-29:
Thanks, it works!

Correct path is `/mnt/exec/release/config/app.json` :)