blikvm/blikvm #238: Request: Support for HTTP/HTTPS Switching

wkumo, 2025-01-15:
HTTPS decryption requires significant machine resources. To enhance the reliability of the server, we suggest providing an option to serve only HTTP. This way, another machine can handle the reverse proxy to offer HTTPS functionality, ensuring better overall system reliability.

m50S79sM6SRNp8Jn, 2025-01-15:
Which BliKVM device hardware you are using?

ThomasVon2021, 2025-01-25:
You can update to v1.5.6 to test

closed 2025-01-25 by ThomasVon2021

slydiman, 2025-02-24:
> You can update to v1.5.6 to test

What is the correct way to switch to http?
I tried to update `/mnt/exec/release/config/app.json` with `"protocol": "http"`. I cannot connect to KVM v4 via http:// after this change. Reverted for now.

slydiman, 2025-02-24:
> Does this solve your problem?

Yes, http works just fine after clearing browser's redirect cache.

m50S79sM6SRNp8Jn, 2025-02-24:
> Yes, http works just fine after clearing browser's redirect cache.

Ok, thanks for confirmation.