blikvm/blikvm #241: No Video Signal (Unable to join Discord, so posting here)

techscx, 2025-01-28:
At first:
Joining discord is not possible. I allways get "Whoops... Unable to accept invite"

**Problem Description**
Hello, I have now Video: "No Signal" after upgrading from 1.4.0 to 1.5.6. Have tried almost anything even installing the full image (1.4.0 release.tar) but that restore disable the whole web interface. After upgrading again (via ssh) to 1.5.6 I ended again in no video signal. Everything else works and there is no error message. Could be that the hdmi port is somehow broken? However, the passthrough is working: My physical monitor connected to the outgoing port is showing the pc screen correctly.

**Proposed Solution**
Video Signal must be working.

m50S79sM6SRNp8Jn, 2025-01-29:
For your problem with Discord, try using incognito browser if you haven't already.
The following links has further suggestions https://thegeekpage.com/unable-to-accept-invite-issue-in-discord-how-to-fix/

For your "no signal" problem. Which BliKVM device did you buy? Show photos of your setup. How target are you trying to control, server, laptop?
If you are in close proximity to your BliKVM, have you tried reflashing using v1.5.6-alpha release?

Please do try to join BliKVM Discord because other users may be able to help you.

ThomasVon2021, 2025-01-30:
Yes, please let me know your hardware version, and upgrading from v1.4.0 to 1.5.6 may have issues. It is recommended to directly re burn the image.

techscx, 2025-01-30:
Hello,

1) discord
The link https://discord.com/invite/ffTDcauqyZ is still not working for me.
"Unable to accept invite"
2) could you provide a step by step procedure to directly flash the latest image.

Thank you

m50S79sM6SRNp8Jn, 2025-01-30:
> 1. discord
>    The link https://discord.com/invite/ffTDcauqyZ is still not working for me.
>    "Unable to accept invite"
> 2. could you provide a ste

https://wiki.blicube.com/blikvm/en/flashing_os/

Which BliKVM device did you buy?

I want to check if by any chance you have been banned. What is your Discord username?
Do you use your Discord username on other Discord servers?
If you are using web browser, then try incognito.
Have you tried Discord on another device?

techscx, 2025-01-31:
Device: It is the BliKVM v4 Allwinner

1.> >  
1.> >  
1.> >  
> https://wiki.blicube.com/blikvm/en/flashing_os/
>
> Which BliKVM device did you buy?
>
> I want to check if by any chance you have been banned. What is your Discord username? Do you use your Discord [...]

ThomasVon2021, 2025-02-01:
> Hello,
> 
> 1. discord
>    The link https://discord.com/invite/ffTDcauqyZ is still not working for me.
>    "Unable to accept invite"
> 2. could you provide a step by step procedure to directly flash the latest image.
> 
> Thank you

https://wiki.blicube.com/blikvm/en/flashing_os/

mggdevit, 2025-02-02:
I encountered the same problem. I connected a brand-new BliKVM v4 Allwinner out of the box. The BliKVM software version was 1.2.0. The remote PC image was displayed in the browser.

I executed the command:

apt-mark hold linux-dtb-edge-sunxi64 linux-image-edge-sunxi64

Then, I updated the system using:

apt-get update && apt-get upgrade

and rebooted.

After that, the image disappeared, showing "No Signal". Possibly, a kernel module got disabled. Reflashing the image onto the memory card (as mentioned above) helped. The new image contains BliKVM version 1.5.2, but the OS is not updated—the root partition is mounted as read-only, preventing updates.

Is it possible to update the OS without breaking the functionality of the BliKVM software?

ThomasVon2021, 2025-02-03:
> I encountered the same problem. I connected a brand-new BliKVM v4 Allwinner out of the box. The BliKVM software version was 1.2.0. The remote PC image was displayed in the browser.
> 
> I executed the command:
> 
> apt-mark hold linux-dtb-edge-sunxi64 linux-image-edge-sunxi64
> 
> Then, I updated the system using:
> 
> apt-get update && apt-get upgrade
> 
> and upgraded BliKVM to version 1.5.6, then rebooted.
> 
> After that, the image disappeared, showing "No Signal". Possibly, a kernel module got disabled. Reflashing the image onto the memory card (as mentioned above) helped. The new image contains BliKVM version 1.5.2, but the OS is not updated—the root partition is mounted as read-only, preventing updates.
> 
> Is it possible to update the OS without breaking the functionality of the BliKVM software?

Do you flash the 1.5.6 image? You can use rw command.

blikvm, 2025-08-04:
Now the latest version is v2.0.4, if this version still has the problem, you can reopen it.