blikvm/blikvm #242: BUG: Paste swallows characters

philkunz, 2025-02-04:
I have blikvm v4 connected to a Mac mini with m4 Pro.

BliKVM seems to swallow the first character when pasting text.

ThomasVon2021, 2025-02-05:
Is it 100% reproducible?

philkunz, 2025-02-05:
It is a 60% occurence, I'd say, that is reproducible across 3 different BliKVMs v4.

Just means there is possibly timing and race conditions involved.

I'm assuming the string is sent as whole to BliKVM? So network should not play a role? Or does the frontend sent each character on its own?

ThomasVon2021, 2025-02-05:
It was sent as a whole. What is the software version and is it the latest 1.5.6?

philkunz, 2025-02-05:
BLIKVM v1.5.6-alpha

m50S79sM6SRNp8Jn, 2025-08-10:
> BLIKVM v1.5.6-alpha

BliKVM v2(alpha) just released, please test it there.

 If there are still problems please send video recording to help use pinpoint the issues and resolve it for you.

Thank you.

m50S79sM6SRNp8Jn, 2025-09-04:
Hi @philkunz 

We have spent quite some effort on our new BliKVM v2 Matrix software (screenshot below).
I would like to know if the issue you faced is now resolved.

When you have a moment please confirm.

[image]

m50S79sM6SRNp8Jn, 2025-09-29:
Hi @philkunz,

v2.1.3-alpha implements further improvement for the issue you reported. 
Kindly reflash to latest version and can you please check?

blikvm, 2026-11-05:
Can you try to update to 3.0.x to test? Very big update.