blikvm/blikvm #244: [Bug][1.5.6] web based user / password manager does not work

arnonm, 2025-02-08:
**Problem Description**
Brand new installation  of CM4.
web login with "admin/admin" does not work, nor with other password

**Proposed Solution**
Describe how you'd like the issue to be resolved and why.

**Alternatives Considered**
Disabling the auth in json file allows login
No change made to default json file

**Additional context**
Error on log file when trying to login:
[2025-02-08T14:00:08.179] [ERROR] file - Error handling HTTP request: SyntaxError: Unexpected token  in JSON at position 0

Json:
{
  "version": 2,
  "log": {
    "console": {
      "enabled": true,
      "level": "info"
    },
    "file": {
      "enabled": true,
      "level": "trace",
      "fileName": "/mnt/tmp/logs/app.log",
      "flags": "a",
      "maxLogSize": 30,
      "backups": 3
    }
  },
  "userManager": {
    "userFile": "./config/user.json"
  },
  "server": {
    "protocol": "https",
    "ssl": {
      "key": "./lib/https/key.pem",
      "cert": "./lib/https/cert.pem"
    },
    "rootPath": "/mnt/blikvm/web_src/web_server",
    "configPath": "/usr/bin/blikvm/package.json",
    "sshUser": "blikvm",
    "sshPassword": "blikvm",
    "auth": true
  },
  "video": {
    "port": 10004,
    "shell": "./lib/kvmd-video.sh",
    "bin": "./lib/pi/ustreamer.bin",
    "fps": 30,
    "quality": 80,
    "kbps": 5000,
    "gop": 30,
    "resolution": "1920x1080",
    "recordPath": "/mnt/tmp/record",
    "recordBin": "./lib/pi/ustreamer-dump"
  },
  "kvmd": {
    "bin": "./lib/pi/kvmd-main",
    "janusBin": "./lib/pi/janus",
    "switch": {
      "enabled": false,
      "devicePath": "/dev/ttyUSB0",
      "module": "BliKVM_switch_v1",
      "list": [
        "BliKVM_switch_v1",
        "BliKVM_switch_v2"
      ],
      "blikvm_switch_v1_lable": [
        "channel_1",
        "channel_2",
        "channel_3",
        "channel_4"
      ],
      "blikvm_switch_v2_lable": [
        "channel_1",
        "channel_2",
        "channel_3",
        "channel_4",
        "channel_5",
        "channel_6",
        "channel_7",
        "channel_8"
      ]
    }
  },
  "display": {
    "isActive": true,
    "mode": 1,
    "onBootTime": 3600,
    "cycleInterval": 60,
    "displayTime": 10
  },
  "atx": {
    "controlSockFilePath": "/var/blikvm/atx.sock",
    "stateSockFilePath": "/dev/shm/blikvm/atx",
    "power_on_delay": 500,
    "power_off_delay": 5000
  },
  "msd": {
    "isoFilePath": "/mnt/msd/user",
    "shell": "./lib/kvmd-msd.sh",
    "stateFilePath": "/mnt/msd/config/msd.json",
    "tusPort": 10002
  },
  "hid": {
    "hidEnable": "./lib/hid/enable-gadget.sh",
    "hidDisable": "./lib/hid/disable-gadget.sh",
    "enable": true,
    "absoluteMode": true,
    "mouseJiggler": false,
    "shortcuts": {
      "Ctrl+Alt+Del": [
        "ControlLeft",
        "AltLeft",
        "Delete"
      ],
      "Alt+Tab": [
        "AltLeft",
        "Tab"
      ],
      "Alt+F4": [
        "AltLeft",
        "F4"
      ],
      "Alt+Enter": [
        "AltLeft",
        "Enter"
      ],
      "Ctrl+W": [
        "ControlLeft",
        "KeyW"
      ]
    }
  }
}

ThomasVon2021, 2025-02-08:
Are you saying that the first time you used the CM4 version was blikvm v1, and then you couldn't log in?
Do you use the v1.5.6 image? Or update to v1.5.6?

arnonm, 2025-02-09:
I reformatted the SD card and installed a new one (as a img file) - v1.5.6
- the latest
SSH works fine - can't logg in to web interface with authentication on .
Error are in the ticket

ThomasVon2021, 2025-02-10:
> I reformatted the SD card and installed a new one (as a img file) - v1.5.6
> - the latest
> SSH works fine - can't logg in to web interface with authentication on .
> Error are in the ticket
> […](#)

So do you use BliKVM v1 or BliKVM v2? Are you using HTTPS to access? Can you join the discord? I need more information to investigate.

ThomasVon2021, 2025-03-03:
Has this issue been resolved？

closed 2025-03-03 by ThomasVon2021