blikvm/blikvm #246: GUI option to add additional users

n3gwg, 2025-02-09:
I would like to see the addition of a GUI option that allows one to add users beyond blikvm and admin.

m50S79sM6SRNp8Jn, 2025-02-09:
duplicate?
https://github.com/ThomasVon2021/blikvm/issues/170

n3gwg, 2025-02-10:
This is not a duplicate at all.

#170 should be bifurcated.

1) Ability to add new users (for whatever reason)
2) The ability for users to have different roles that are granular enough to impact which channel they are going to be administering.

blikvm, 2026-10-28:
Can you try to update the v3.0.1 image. It support this function now. https://github.com/blikvm/blikvm/releases 
Much better than v2.x.x version. You can also play first https://blikvm.com/demo/