blikvm/blikvm #247: GUI option to add additional users

n3gwg, 2025-02-09:
It could be useful to plug an external camera (to see the server rack area) into the BliKVM, can an option be added to support showing the video from such an external camera?

ThomasVon2021, 2025-02-10:
Great suggestion, we will consider it. A USB camera, maybe.

n3gwg, 2025-02-10:
I was entirely thinking of a USB camera! Perhaps a common type that would be easily supportable and inexpensive.

I am not trying to take live video of atoms moving about, just want to see my rack or some status lights on other items in the rack, that sort of thing.

ThomasVon2021, 2025-02-11:
We will add this feature in future versions.

blikvm, 2026-10-28:
Can you try to update the v3.0.1 image. It support this function now! https://github.com/blikvm/blikvm/releases 
Much better than v2.x.x version. You can also play first https://blikvm.com/demo/

blikvm, 2026-11-05:
If you try this function, still have problem, you can reopen this issue again.
https://blikvm.com/docs/video/cameras

closed 2026-11-05 by blikvm