blikvm/blikvm #249: Keyboard|mouse connection error

vpaulos, 2025-02-15:
**Problem Description**
Briefly explain the issue or need for the feature. For instance, "It would be helpful if [...]" or "I often experience challenges when [...]".

**Proposed Solution**
Describe how you'd like the issue to be resolved and why.

**Alternatives Considered**
List any other solutions or approaches you've thought about.

**Additional context**
Provide any extra details, references, or screenshots that might help explain your idea.

No Key|mouse connected to remote device.
What it can be?
All wires is connected.

[image]

ThomasVon2021, 2025-02-16:
1. Are you sure the USB cable supports data transfer?
2. Is it a directly burned v1.5.6 image?
3. show me the /boot/config.txt

m50S79sM6SRNp8Jn, 2025-02-16:
Which BliKVM device do you have?

vpaulos, 2025-02-17:
Hi! I`ve BliKVM v1 CM4 "KVM over IP" Raspberry Pi CM4 HDMI CSI PiKVM v3.
yes I burn  v1.5.6 image to the flash.
 it is running and I can see "video" from remote host, but keyboard and mouse is not active and do not installed into remnote host. I try to use both cables: type-c <-> type-c and type-c <-> usb 2.0 keyb&mouse is inactive in remote host.

next step i find some issue: if after booting I going into terminal and mannually run ./diasble script and after that enable sh script - key board and will working, but until next reboot or restart!

ThomasVon2021, 2025-02-17:
Which  diasble script  and  enable sh?

vpaulos, 2025-02-17:
ok. will give you full path. it was in ../hid/ folder and name like ./enable-****.sh  that,

ThomasVon2021, 2025-02-18:
show me the /boot/config.txt
Also, when are you available? Can we investigate together online, such as through Discord. I am in Beijing time, so we need to consider the time difference between us. Another troubleshooting issue is whether it is a software problem, which is to flash pikvm images for testing.

ThomasVon2021, 2025-02-18:
I burned the v1.5.6 image today and did not find any issues during testing. It may be necessary to analyze it in conjunction with your device.

m50S79sM6SRNp8Jn, 2025-02-18:
> Hi! I`ve BliKVM v1 CM4 "KVM over IP" Raspberry Pi CM4 HDMI CSI PiKVM v3. yes I burn v1.5.6 image to the flash. it is running and I can see "video" from remote host, but keyboard and mouse is not [...]

Ensure you are using a data cable (power cable will not work). This is usually the problem.
In case you are using any splitter remove it. Why? To keep it simple.

If this doesn't' fix your problem send photos of your setup.

The best way to get help is to join BliKVM discord, https://discord.gg/9Y374gUF6C

closed 2025-02-24 by m50S79sM6SRNp8Jn

vpaulos, 2025-02-28:
/mnt/exec/release/lib/hid/disable-gadget.sh

vpaulos, 2025-02-28:
cable is ok. if after power on go to console and put
rw
cd /mnt/exec/release/lib/hid/
./disable-gadget.sh
./enable-gadget.sh

all will be ok and key and mouse will available

m50S79sM6SRNp8Jn, 2025-02-28:
Do you mean the problem is resolved?