blikvm/blikvm #258: [Bug Report] [v1.5.6-alpha] Keyboard|mouse connection error #249

vpaulos, 2025-02-28:
**Problem Description**
Briefly explain the issue or need for the feature. For instance, "It would be helpful if [...]" or "I often experience challenges when [...]".

**Proposed Solution**
Describe how you'd like the issue to be resolved and why.

**Alternatives Considered**
List any other solutions or approaches you've thought about.

**Additional context**
Provide any extra details, references, or screenshots that might help explain your idea.

double for closed [Bug Report] [v1.5.6-alpha] Keyboard|mouse connection error #249

vpaulos, 2025-02-28:
cable is ok. if after power on go to console and put
rw
cd /mnt/exec/release/lib/hid/
./disable-gadget.sh
./enable-gadget.sh

all will be ok and key and mouse will available

m50S79sM6SRNp8Jn, 2025-02-28:
If the issue is resolved, please close it, thanks.

ThomasVon2021, 2025-03-01:
Are you referring to the situation where the keyboard and mouse cannot work when kvm automatically starts up. Do you have to disable first and then enable before kvm can work?
If that's the case, I think I may need to schedule a time with you to remotely access your device in order to see the possible cause. I am unable to reproduce this issue myself.

vpaulos, 2025-03-04:
yes, situations is always that. after reboot or power on, i have this problem always, I need to restart manually service via terminal script and all works fine. but if device will restart - all the same -> disbale&&enable scripts

vpaulos, 2025-03-04:
what logs can I send you from this device?

ThomasVon2021, 2025-03-04:
/mnt/tmp/logs/app.logs
You can first clear all the logs, then restart the device, open the web, if the mouse cannot write, and then send me this app.log to have a look.

ThomasVon2021, 2025-03-04:
[image]

blikvm, 2026-10-28:
Can you try to update the v3.0.1 image. https://github.com/blikvm/blikvm/releases 
Much better than v2.x.x version. You can also play first https://blikvm.com/demo/