blikvm/blikvm #285: [Bug Report] [v1.6.0-alpha] blikvm v4 the screen cannot be displayed after the update.

koohuang, 2025-05-19:
After updating from version 1.5.9-alpha to 1.6.0-alpha, the screen no longer displays

blikvm, 2025-05-19:
```
systemctl stop kvmd-web
cd /mnt/exec/release/
./server_app
```
show me the output.

koohuang, 2025-05-19:
root@mangopimcore:/mnt/exec/release# ./server_app
Config path is: config/app.json and switch path is: config/switch.json
[2025-05-19T14:17:17.871] [INFO] console - app.json file successfully update!
[2025-05-19T14:17:17.891] [INFO] console - user.json file successfully update!
[2025-05-19T14:17:17.898] [INFO] console - switch.json file successfully update!
[2025-05-19T14:17:17.904] [INFO] console - wake_on_lane.json file successfully update!
[2025-05-19T14:17:18.161] [INFO] console - Https Api started at https://localhost:443, state: STARTING
[2025-05-19T14:17:18.165] [INFO] console - WebSocket Api started at ws://localhost:443, state: STARTING
[2025-05-19T14:17:18.166] [INFO] console - tus server listening at http://0.0.0.0:10002
[2025-05-19T14:17:18.175] [INFO] console - HID already running
[2025-05-19T14:17:18.271] [INFO] console - All services have been started.
[2025-05-19T14:17:18.288] [WARN] console - Notification WebSocket is not initialized.
[2025-05-19T14:17:18.415] [INFO] console - video API started, state: RUNNING
[2025-05-19T14:17:18.439] [INFO] console - kvmd-main API started, state: RUNNING
[2025-05-19T14:17:18.468] [INFO] console - janus API started, state: RUNNING
[2025-05-19T14:17:19.366] [INFO] console - video API exited with code 127 and signal null
[2025-05-19T14:17:19.370] [INFO] console - video API closed with code 127 and signal null
[2025-05-19T14:17:20.596] [INFO] console - video API started, state: RUNNING
[2025-05-19T14:17:21.491] [INFO] console - video API exited with code 127 and signal null
[2025-05-19T14:17:21.495] [INFO] console - video API closed with code 127 and signal null
[2025-05-19T14:17:22.722] [INFO] console - video API started, state: RUNNING

blikvm, 2025-05-19:
you can try to install 
```
apt install libx264-dev libyuv-dev
```
If still have problem, you can show me this command output
```
./lib/h616/ustreamer.bin --format=MJPEG --encoder=LIBX264-VIDEO --device=/dev/video1 --resolution=1920x1080 --host=0.0.0.0 --port=10004 --drop-same-frames=30 --desired-fps=30 --h264-bitrate=5000 --h264-gop=30 --h264-sink=demo::ustreamer::h264
```

koohuang, 2025-05-19:
> you can try to install
>
> ```
> apt install libx264-dev libyuv-dev
> ```
>
> If still have problem, you can show me this command output
>
> ```
> ./lib/h616/ustreamer.bin --format=MJPEG --encoder=LIBX264-VIDEO [...]

It displays correctly after installation. Thank you for your support

closed 2025-05-19 by blikvm

dmitriyteteruk, 2025-05-19:
Had same issue after update.