blikvm/blikvm #290: Mouse Jiggler

kleverbatista, 2025-08-01:
**Problem Description**
was disabled on last version 2.0.3
**Proposed Solution**
Add mouse jiggler again

m50S79sM6SRNp8Jn, 2025-08-02:
[image]

blikvm, 2025-08-02:
[image]
You can view the latest logs in this path and confirm if there are any related prints in the logs

kleverbatista, 2025-08-02:
Perfect, I will use 60 seconds to be sure it is active, remember to add the virtual mouse button too.

blikvm, 2025-08-03:
Can you show me what a virtual mouse button looks like? Any example.

kleverbatista, 2025-08-03:
On Android and iOS devices, when using the web portal of BliKVM, the touch input correctly positions the cursor on the screen. However, there is currently no support for left-click or right-click actions through touch, making it impossible to interact with the interface unless a physical mouse is connected.

A useful improvement for the WebUI would be to include virtual mouse buttons—similar to what PiKVM already implements—so users can send left and right clicks directly from the interface.

For example, when using an iPad with a keyboard and trackpad, everything works fine, including clicking. But on an iPhone or any other touchscreen device without a physical mouse, there is no way to send click actions. Adding on-screen buttons for left and right clicks would solve this limitation and make the system usable from any mobile device.

[image]

blikvm, 2025-08-03:
Thank you very much for the detailed screenshot. Can you add a virtual mouse and open a separate issue.

blikvm, 2025-08-03:
Try to update to v2.0.4.

kleverbatista, 2025-08-03:
> Thank you very much for the detailed screenshot. Can you add a virtual mouse and open a separate issue.

#292

blikvm, 2025-08-04:
You can update to v2.0.4.

blikvm, 2025-08-08:
Do you test it?