blikvm/blikvm #292: Can you add a virtual mouse

kleverbatista, 2025-08-03:
On Android and iOS devices, when using the web portal of BliKVM, the touch input correctly positions the cursor on the screen. However, there is currently no support for left-click or right-click actions through touch, making it impossible to interact with the interface unless a physical mouse is connected.

A useful improvement for the WebUI would be to include virtual mouse buttons—similar to what PiKVM already implements—so users can send left and right clicks directly from the interface.

For example, when using an iPad with a keyboard and trackpad, everything works fine, including clicking. But on an iPhone or any other touchscreen device without a physical mouse, there is no way to send click actions. Adding on-screen buttons for left and right clicks would solve this limitation and make the system usable from any mobile device.

[image]

blikvm, 2025-08-13:
A test demo.
https://github.com/user-attachments/assets/75508e0d-4584-40ab-9ab7-cdfc9ddd5a7e

kleverbatista, 2025-08-14:
That's awesome!! Hope you can release it soon!!

closed 2025-08-20 by blikvm