blikvm/blikvm #303: Add nut to blikvm to facilitate shutdown on power outage

HappyPants9999, 2025-08-16:
**Problem Description**
Briefly explain the issue or need for the feature. For instance, "It would be helpful if [...]" or "I often experience challenges when [...]".

**Proposed Solution**
Describe how you'd like the issue to be resolved and why.

**Alternatives Considered**
List any other solutions or approaches you've thought about.

**Additional context**
Provide any extra details, references, or screenshots that might help explain your idea.

HappyPants9999, 2025-08-16:
Would like to see nut client/server added to blikvm in order to shutdown everything cleanly on power outage.

HappyPants9999, 2025-09-10:
> I was able to put client on, but native support for either client or server would be great a great feature.
>
> Please give me details in the client. Did you use any npm package?

I used apt to install nut.

blikvm, 2026-11-05:
Hello, do you still have this problem. Now you can update to 3.0.x, we have a big update.